furbrain/STIC #8: Document production process

furbrain, 2024-01-08:
- [ ] Make BOM
- [ ] link to files on github releases
- [ ] update docs

milewis, 2024-09-03:
Phil, 

I'm working on building a SAP from scratch.  I've got the board soldered up and have the device showing as circuitpy on my computer.  I'm a bit lost as to which files I should be moving over to the nrf device.  Any direction would be very helpful.  I'd also love to help with any documentation I can.  Are you needing anything on the build process documented in particular?

furbrain, 2024-09-07:
The first thing is you need a special version of circuitpython installed - you can get it from https://github.com/furbrain/SAP6_board/releases/download/1.0/sap6_cp_9.0.0.uf2
You will need to create a `firmware` directory on the nrf device and put all the python files from `firmware.zip` in there. 

You will also need to create and populate the `fonts` and `images` directories, and also copy `safemode.py` `boot.py` and `code.py` from `firmware/installer` into the root directory. The final directory structure should look like:
```
├── boot.py
├── code.py
├── firmware
│   ├── ***lots of python files***
│   └── versions
│       ├── display128x64.py
│       ├── hardware_v1.py
│       └── __init__.py
├── fonts
│   ├── limited
│   ├── terminus_20_ascii.pcf
│   ├── terminus_22_ascii.pcf
│   ├── terminus_24_ascii.pcf
│   └── terminus_28_ascii.pcf
├── images
│   ├── battery.bmp
│   ├── batt_icon.bmp
│   ├── bt.bmp
│   ├── laser.bmp
│   └── laser.xbm
├── manual.pdf
└── safemode.py
```