furbrain/STIC #20: consider power-off with double-click B instead of A

echarlie, 2024-02-06:
On a survey trip with the SAP6 over the weekend, I found we would shut down the SAP on accident fairly often, either by muscle spasm while taking a shot, or it not responding immediately when turning on (so you get a trill of the power on, then the power off sound, before the screen has powered up).

Changing power-off to the B button would ensure this happens accidentally less often.

furbrain, 2024-02-07:
Unfortunately double-click B is now taken as "return to most recent reading", when browsing through previous readings. 
There is a lockout to prevent turn-on- turn-off, but it's only 0.5s. I'd be open to a PR to change this to say 2-5 seconds (it's in quitter_task in app.py)

Phil

echarlie, 2024-02-12:
"return to most recent reading" sounds better. I was sketching out a bit of code to allow going back and forth through old readings, but I think that is 1) more memory-intensive, and 2) creates a very weird behaviour for exiting to the measure mode. (see https://github.com/furbrain/STIC/commit/26f3e01bdbf80b00d19f56799c10245ebc8a9d1a if you're curious).