furbrain/STIC #25: Unclear PCB instructions

mlubej, 2024-05-13:
Hi!

Thank you for this really nice project, I love the fact that you took the time to make it so open-source and enable us to build our own.

I will try to go through the process of creating my own device, but have limited experience with hardware.

I just ordered the manufacturing of the PCBs on pcbway.com and I have received a few question from their side regarding the files, such as:

First question:

> Pls check the attached pic,the pointed pad of the top copper layer is covered with soldermask ink.So do you need us to add soldermask opening for the pointed pad to make it exposed?

[image]

Second question is very similar:

> Pls check the attached pic,the two circled pads are also both covered with soldermask ink.So do you need to add soldermask openings for the two pads to make them exposed？

[image]

Am I correct in assuming that the marked parts should have an opening in the solder mask ink?

Thank you!

PS: I might ask a bit more often about the manufacturing process and try to document any issues so that we can improve the docs/manual.

milewis, 2024-05-13:
None of those pads are used in the design, so they have been covered up by solder mask in the most recent version of the board.  The only pad that is under the chip PCB is the BATT pad, which has a hole through to the back side so that it can be easily soldered with a torch.

furbrain, 2024-05-13:
Exactly as milewis has said - these are deliberate design decisions.

Phil

furbrain, 2024-05-13:
Matic, thank you for your help with this please do identify any issues that
you come across!

Phil

mlubej, 2024-05-17:
Thank you for the info!

mlubej, 2024-05-17:
BTW, due to my lack of knowledge I ordered them to create openings in the solder mask ink before asking you here. Do you think this is a problem? I guess I can cover the parts myself?

furbrain, 2024-05-17:
I think it is unlikely to be a problem. Previous versions if the board did
not have the solder mask and it was fine.

Phil

mlubej, 2024-05-17:
Thanks for the quick reply! Looking forward to mowing forward with this.

closed 2024-05-17 by mlubej