furbrain/STIC #26: Ordering components

mlubej, 2024-05-17:
Hi @furbrain!

I wanted to order the electrical components, but realized that the purchase might need to be quite dispersed across various sites, especially for the more complex components, such as the magnetometer, BLE, laser module, etc... I have a few questions:

- do you order the laser module directly from Egismos?
- where do you prefer to order the items from the `bom.csv` which don't have the Farnell part number?
- in the BOM there are 6 resistors in groups of 3 which seem to be different, but point to the same Farnell part number (9342788). Is this a mistake?
- how do you manage to sell the SAP6 at such a small price compared to the sum of costs I need to pay for the components alone? 😂 

Thank you!

furbrain, 2024-05-18:
- Yes I order the laser modules directly from egismos. There was a time when they were available from Amazon, but no longer
- The items which are not available via farnell I tend to get from aliexpress - there are urls in bom.csv. You don't need U4
- The 6 resistors with the same part number is an error. There are two lots of resistors and capacitors, depends on whether you want to use through hole or surface mount. So you either need R1,R2,R3 (SMT) *or* R_1, R_2, R_3. Any 2k resistor in 0603 package is suitable for SMT. The same applies for C1, C_1 etc
- I get stuff cheap when I gets lots of them (especially with the laser module - cost of shipping per unit is much better hen buying in bulk)!

mlubej, 2024-06-17:
Hi!

Thank you for your reply, it was helpful. I managed to order the Laser and some of the other components. I might drop in with some question every now and then and I'll use this ticket here.

- I think I spotted an error with the manufacturer number for `C1` and `C2` components. The stated number is the same as for the `C3` capacitor, but this can't be, since the capacitance is different. Perhaps [this one](https://www.aliexpress.com/item/1005006523977750.html?spm=a2g0o.cart.0.0.584338da33Svnz&mp=1) should be stated?
- I'm having trouble understanding the `U4` component. It seems to be linked to the RM3100 breakout board (`U2` component), but it's not exactly clear what this component is, nor can I find it due to missing part numbers. can you please help me out here?
- for the `J4` component I see that the 100mm long cable is used. do you think the 150 mm cable would fit as well?

furbrain, 2024-06-17:
Yes, there is an error in my parts list - I will update shortly. The part you have identified is appropriate, my device uses https://jlcpcb.com/partdetail/16531-CL10A105KB8NNNC/C15849. However, the 4.7uF capacitors would probably work perfectly well.

The J4 jumper cable - 100mm works very nicely, you would likely get away with the 150mm but might need to tuck it out of the way of the laser once assembled.

U4 was sort of an experiment - there is a different breakout board for the RM3100, which it looked like I might need to use; I was going to try to fit both options on the board but it didn't work out.

Do let me know if any more questions!

mlubej, 2024-10-05:
Hi @furbrain!

Progressing well with building the SAP! I have a few more questions, mostly about the battery

- turns out the battery I received has 2 wires instead of the 3  like in case of the one in the `bom.csv`. Do you think that's a problem?
- in the `bom.csv` there is the `J5` component (the JST 2 way header) as well as the `J1` component (molex 3 way pin header), which both seem to be going to the same place on the PCB. Does that mean that you made a design so both kinds of batteries can be used, or am I doing something wrong?
- for the laser, I have received a 5-way cable which connects to the laser module, but has wires on the other side, so I'm missing a 5 way single row female pin header which I can plug in the board. Was this also in your case? Is the component missing from the `bom.csv`?

furbrain, 2024-10-05:
Hi @mlubej 

- Two wires on the battery is no problem, but you will need to use the J5 component, not the J1. Make certain you get the polarity correct
- Yes the design was meant so you could use either type of battery
- I think there is an error in the bom - you will need a 5 way picoblade cable with female headers on both ends - you will need part number 15134-0500