furbrain/STIC #27: Battery question - is it non-magnetic?

mlubej, 2024-07-12:
Hi @furbrain!

One question regarding the battery. The Disto-X2 used for caving uses the non-magnetic [LiPo battery (PGEB-NM053040)](https://www.powerstream.com/non-magnetic-lipo.htm), but the [battery used in SAP6](https://cpc.farnell.com/unbranded/lp-523450-is-3/battery-lithium-pol-3-7v-950-mah/dp/BT06802) doesn't seem to have any specific information regarding the battery. Do you have any information if the current battery is non-magnetic, or if the draining currents of the battery may affect the magnetometer and therefore the measurements?

Comparing the batteries, it seems that both are 3.7 V and 5 mm thickness, and the non-magnetic one is even smaller in width x height. Does this mean that the battery could be changed basically in place?

Thank you!

furbrain, 2024-07-12:
I've tested various lipo batteries with the magnetometer, and none of them
have any significant ferromagnetism to the extent that it affects the
readings.

Any electrical current will of course generate a magnetic field which is
why I have tried to keep the battery away from the magnetometer.

The battery can be exchanged for a different one with no problems, so long
as it is at least 500mAh. Smaller batteries will have problems because the
charging current would be too high

mlubej, 2024-07-15:
Thank you so much for this information!

closed 2024-07-15 by mlubej