furbrain/STIC #28: SAP6 feedback - physical attributes

mlubej, 2024-07-12:
Hi @furbrain!

At our local caving club we have received the SAP6 and have already field-tested it. There were some changes in measurements compared to the Disto-X2, but the Disto-X2 was not calibrated, so we have to repeat this test. Otherwise the calibration process of the SAP6 is really simple and the device is intuitive enough. We may change the interface and button usage a bit to fit more to our needs. If there are any generic updates or bugfixes, we will prepare a PR.

There were a few weird crashes which we cannot reproduce yet, but we'll try it out more. We even have all the electrical components now to try to make our own (except for the battery, because we'd like your input from #27 first)

One comment from our side is that the 3D prints seem a bit bad, and for a purchased item such as this we would have expected better quality prints. Perhaps your 3D printer is not calibrated well. 

The print seems to be warped at the base and at the screen part

|||
|-|-|
|[image]|[image]|

and the cap has some holes in the layers, which is a sign of under-extrusion

[image]

Additional comment regarding the glass layer: is there a way that we could protect this? did you have any ideas in the past already? If you agree that it would be a good idea, I can design a TPU "cap" with a hole which would fit on the glass part. Accepting also other ideas

[image]

Will continue testing and giving feedback, thanks!

mlubej, 2024-07-12:
BTW, I see that you used ABS for this, which is famous for warp issues on the print bed. IMO ABS is best for situations where high-temperature resistance is needed, but if that is not a requirement, PET-G is equally good, since it's still very robust and much easier to print. Do you think it would be a suitable replacement to ABS in this product?

furbrain, 2024-07-12:
> BTW, I see that you used ABS for the prints, which is famous for warp

Unfortunately the reel of filament for your device did have some warping. I
have now moved to a commercial 3d printer and have not had any further
issues with it

furbrain, 2024-07-12:
About the glass part: this is laser cut acrylic, so it's fairly robust. It
would also be easy to replace if it became damaged. Currently the device
fits precisely in a peli case, so I don't want to make it any bigger for
the general user. I think having an optional tpu protector could be a good
idea however, maybe I could have this as an accessory. Also spare front
pieces cloud be an idea

furbrain, 2024-07-12:
Other people have used some clear adhesive tape on the front of the device
to use as sacrificial protection which can then be really replaced