furbrain/STIC #31: Alternate laser module - TW10S-UART

milewis, 2024-11-08:
I'm looking at using an alternate laser module that is available very inexpensive on Aliexpress for $30.  The TW10S-UART  uses UART and will connect to the same 5-pin port that the Egismos connects to on the SAP board, but uses different pin locations (I have just moved the wires in the wiring harness for testing.  I am currently working on a port of the laser_Egismos library to change the packet data from the commands for the Egismos to the commands for the TW10S-UART.  The downside I have seen so far is that the distance is lower, but  for the caves I am currently surveying, this is not an issue as my shots are rarely over 50' (15m).  I have added the documentation from the site where I purchased the laser module that goes over the UART commands.
[TW10S-UART Documentation.docx](https://github.com/user-attachments/files/17683242/TW10S-UART.Documentation.docx)

furbrain, 2024-11-08:
That looks very interesting! I like that it has a proper connector on it
(other laser modules need to have wires soldered directly on to the board).
My experience is that the maximum range is only achievable in perfect
conditions - no mist, matt white target etc. I've found in caves with dark
shiny rock you often get a much shorter maximum range. With the sap5, using
a cheaper 30m laser I was getting many people reporting poor performance.

It should be reasonably easy to generate a new hardware version.

I'll put an update to the docs with me details on how to implement a new
hardware version

furbrain, 2024-11-08:
Of course the best thing to do is test the lasers in real life
conditions....

furbrain, 2024-11-10:
I've put some more information about making ponies with different hardware [on readthedocs](https://stic.readthedocs.io/en/latest/#make-a-derivative)