furbrain/STIC #37: Bluetooth Not Found with SAP6 1.5.0

BenMorganMY, 2025-08-19:
Hey there,

I grabbed a Samsung Galaxy A16 yesterday and went to connect the SAP6. I have not been able to see the device show up on the Bluetooth. I have been having other devices search for it and they are not able to see the SAP6. I have tried resetting, Bluetooth > Disconnect, and Bluetooth > Forget Pairings to no avail.

I'm wondering if there's something I'm missing to set the SAP6 into a pairing mode. It has been previously paired with 1 other phone before.

furbrain, 2025-08-20:
Are you trying to connect from there phone Bluetooth settings screen or
from survey software? If so, which software ( and version) are you using?

Phil

furbrain, 2025-08-20:
> When I get home, I'll get that information for you. For now, I got this
> fixed by downloading SexyTopo and letting it do the bluetooth scan. Then it
> picked up the SAP6_JP and connected it. When I went over to TopoDroid is
> was then able to discover and use it. Very odd, but it appears SexyTopo did
> something TopoDroid did not.

Thanks for the feedback - sounds like this may then be a topodroid software
issue.

Closing as doesn't seem to be a SAP6 hardware issue, but happy to reopen if
more info

Phil

closed 2025-08-20 by furbrain

BenMorganMY, 2025-08-20:
Perhaps, I did find it interesting that I could not find `SAP_JD` in the list of devices to connect to in the Bluetooth Connections directly and instead needed SexyTopo to find it for me. As a user, I would expect to go to my bluetooth settings, find the device, then head to my app and request the app to use the device.